jonblack/arduino-menusystem #12: ESP8266

fazioa, 2015-09-23:
It not work on ESP8266! 
:(

jonblack, 2015-09-23:
Can you be more specific? What error are you seeing? What code are you trying to run? I'd never heard of an ESP8266 until now, so it's difficult for me to help without a clear definition of the problem.

fazioa, 2015-09-23:
I'm sorry.
Close please.
It compile with warning but work fine!

It compile with message:
ATTENZIONE: la libreria arduino-menusystem dichiara di funzionare sulle architetture [avr] e potrebbe essere incompatibile con la tua attuale scheda che utilizza l'architettura [esp8266].

current_item.ino:14:11: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
current_item.ino:15:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
current_item.ino:16:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
current_item.ino:17:35: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
current_item.ino:18:43: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
C:\Users\Antonio\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\cores\esp8266\libc_replacements.c:408:22: warning: 'struct tm' declared inside parameter list [enabled by default]
 time_t mktime(struct tm *timp) {
                      ^
C:\Users\Antonio\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1106-g8253b82\cores\esp8266\libc_replacements.c:408:22: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]

closed 2015-09-23 by fazioa