jonblack/arduino-menusystem #13: Warning during compile in 1.6.6

TurBoss, 2015-11-24:
HI,
I get this messages when I compile the examples

```
arduino-menusystem\examples\serial_nav\serial_nav.ino:14:26: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Menu mm("ROOT Menu Title");
^
arduino-menusystem\examples\serial_nav\serial_nav.ino:15:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
MenuItem mm_mi1("Level 1 - Item 1 (Item)");
^
arduino-menusystem\examples\serial_nav\serial_nav.ino:16:42: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
MenuItem mm_mi2("Level 1 - Item 2 (Item)");
^
arduino-menusystem\examples\serial_nav\serial_nav.ino:17:35: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
Menu mu1("Level 1 - Item 3 (Menu)");
^
arduino-menusystem\examples\serial_nav\serial_nav.ino:18:43: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
MenuItem mu1_mi1("Level 2 - Item 1 (Item)");
^
```

thx

jonblack, 2015-11-25:
This looks like a minor bug. The library classes all use a `char*` whereas `"hello"` is a `string` constant. It's being converted automatically and apparently that's now deprecated (perhaps the `write-strings` compiler flag was recently added?).

It's a warning so you can continue (you probably know this already). I'll work on a fix for this soon. I'm a bit swamped at the moment. Thanks for reporting the problem. I appreciate it.

jonblack, 2016-01-11:
Fixed in master. Will end up in a tagged release soon so its picked up by the library manager.

closed 2016-01-11 by jonblack

TurBoss, 2016-01-11:
ty