jonblack/arduino-menusystem #14: Reset menu

firebull, 2016-02-03:
What is the way to reset menu? I mean if I close it, so when I open it again it is opened on the last item.

So how can I make it to start from the begining?

jonblack, 2016-02-03:
That depends on how you close it. If you `select` and item, its parameter `return_to_root`, which defaults to `true`, will reset the menu. If nothing is being selected then there currently isn't a way to do this. Does this describe your use case?

firebull, 2016-02-03:
I have an item named "Exit" which executes the function:

```Arduino
void close_menu(MenuItem* p_menu_item){
  menuSelected    = false;
  commandReceived = false;
  adcKeyPressed = -1;
  lcd.clear();
}
```

So then the other question: what is the correct way to close the menu?

jonblack, 2016-02-03:
I understand that you have a menu and one of those items is **Exit**. When this item is selected, the above callback is called.

When does your code call `select`? Also, what's the purpose of `menuSelected`?

firebull, 2016-02-03:
Well, I translate analog read to button and then run the function:

```Arduino
void controlAction(){
  switch(adcKeyPressed){
    case 0:
      ms.prev(); 
      displayMenu();
      break;
    case 1:
      ms.next();
      displayMenu();
      break;
    case 2:
      ms.select();
      break;
    case 3:
      ms.back();
      displayMenu();
      break;  
    default:
      break;   
  }

  adcKeyPressed = -1;
  commandReceived = false;
  return;
}
```

The purpose of menuSelected to define what to show: basic work screen or menu.

jonblack, 2016-02-03:
Since `return_to_root` defaults to `true`, the menu should reset. Can you show me the code that builds the menu. This is typically done in `setup`.

firebull, 2016-02-03:
```Arduino
MenuSystem ms;
Menu mm("Menu");
MenuItem mm_mi1("Keeping");
MenuItem mm_mi2("Power");
MenuItem mm_mi3("Screw feed");
MenuItem mm_mi4("Exit setup");
Menu mu1("Date & Time");
MenuItem mu1_mi1("Set");
MenuItem mu1_mi2("Go back");

void setup(){  
  mm.add_item(&mm_mi1, &keep_setup);
  mm.add_item(&mm_mi2, &power_setup);
  mm.add_item(&mm_mi3, &screw_setup);
  mm.add_menu(&mu1);
  mu1.add_item(&mu1_mi1, &on_time_set_selected);
  mu1.add_item(&mu1_mi2, &on_back_selected);
  mm.add_item(&mm_mi4, &close_menu); // Menu close to the end
  ms.set_root_menu(&mm);
}
````

jonblack, 2016-02-03:
I'm sorry to say this but it's a bug. The `return_to_root` aspect of `select` doesn't even come close to resetting the menu. What it does do is set the current `Menu` to the root, but does nothing to reset the current selected `MenuItem` within a `Menu`.

I also wonder if it's useful having this functionality on `select`. Since `select` is often called in a loop, there's no fine-grained control, which I imagine some users may want (e.g. some selections cause a menu reset, others don't).

I suspect we need a `reset` option on the `MenuSystem` class which goes through the menu properly to ensure not only that `_p_curr_menu == p_root_menu` but also that `_cur_menu_component_num == 0` in a `Menu`.

So for now I don't think there's much you can do. I'll try and fix this as soon as possible, but that might be a couple of weeks as I'm quite busy at the moment.

commit 2d729e28 (2016-03-20): Fix menu reset bug

`select` can no longer be used to reset the menu. Instead, a separate
`reset` function is provided.

Fixes #14
commit 1c86e29b (2016-04-22): Fix menu reset bug

`select` can no longer be used to reset the menu. Instead, a separate
`reset` function is provided.

Fixes #14
closed 2016-03-20 by jonblack