jonblack/arduino-menusystem #16: Suggestion

peekpt, 2016-03-06:
Hi,

I know it's possible to achieve this in code, but for current item indicator is it possible to add like a cursor char prefix to the name of the items  when we call the getName function?
Because ,concatenate  pointers it's a little tedious.
Something like:  set_cursor(const char * prefix, const char* suffix);

jonblack, 2016-03-06:
By cursor you mean the ASCII character used to show the current selected item, right?

The library is intended to be generic. So whilst it's possible to build a menu where the current item has a prefix to show it's selected, that's not always the case. In fact, I often show just the current menu item. So there are a lot of cases where a cursor doesn't make sense. For that reason, I don't think it's beneficial to add a `set_cursor` function.

peekpt, 2016-03-06:
Thank you very much for your quick reply. 
Yes, that's right and I totally understand your point.
In my case, I'm building a menu for the 128x64 oled display using u8glib like this:

Menu
——————
> Item1
   Menu1*
   Item2
   Item3

It's already working fine but I will like to tune it a little bit.

jonblack, 2016-03-08:
During your menu construction you could loop over the menu components and if the menu component in the current loop is the selected one, apply the cursor only for that item.

Are you happy for me to close this as not an issue?

peekpt, 2016-03-08:
Yes please you.may close it. thank you!

closed 2016-03-08 by jonblack