jonblack/arduino-menusystem #17: 2.1.0 release - undefined reference to `MenuSystem::select()'

hestenet, 2016-03-21:
After recompiling my code following the 2.1.0 release I'm seeing a large number of 
`undefined reference to `MenuSystem::select()'` errors.''

I'm not sure why... could it have something to do with this change:
` MenuSystem.h line 82 
-    void select(boolean return_to_root=true);`

select() no longer as a return_to_root behavior?

jonblack, 2016-03-21:
I removed `return_to_root` and replaced it with a separate function `reset`. In retrospect perhaps that wasn't the best idea. I was looking at it from the use case where a single menu item causes the reset, but some people may want to reset after every call to `select`. This is possible with `reset` but would require a call to reset in all callbacks, which is a bit cumbersome.

I'll reintroduce the option, though it'll be renamed: `MenuComponent* select(reset=false)`.

hestenet, 2016-03-21:
That sounds good! I'll be happy to test it out.

jonblack, 2016-03-22:
This is now fixed in 2.1.1.

closed 2016-03-22 by jonblack

hestenet, 2016-03-23:
Thank you - my sketch from 2.0.2 now compiles on 2.1.1 without errors.