jonblack/arduino-menusystem #21: Display

recyclix, 2016-04-04:
Hi Jon,

I would like to display the setpoint value for each item on the SETTINGS> menu while scrolling the menu.
So an operator can see the setpoint value without selecting each item individually.

`// include the library code:
#include <LiquidCrystal.h>
LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

//#include <Adafruit_RGBLCDShield.h>
//#include <utility/Adafruit_MCP23017.h>
//// Initiate Display
//Adafruit_RGBLCDShield lcd = Adafruit_RGBLCDShield();
//#define RED 0x1

#include <MenuSystem.h>
// Menu variables
MenuSystem ms;
Menu mm("");
MenuItem mm_mi1("HIGH ");
MenuItem mm_mi2("MID");
MenuItem mm_mi3("LOW");
MenuItem mm_mi4("FREE");
Menu mu1("SETTINGS>");
MenuItem mu1_mi1("HighSp");
MenuItem mu1_mi2("MidSp");
MenuItem mu1_mi3("LowSp");
MenuItem mu1_mi4("SlowDownSp");
MenuItem mu1_mi5("IgnDelaySp");
MenuItem mu1_mi6("AlarmDelaySp");

enum setMenuSelected_Type {
  Settings, Other
}; // this enum is in case we want to expand this example
setMenuSelected_Type setMenu;
boolean menuSelected = false;

int Setpoints[] {1000, 750, 500, 100, 80, 10};
int SpSelected;

void loop() {
  serialHandler();
} // end loop

//Menu operations
void on_SuspHi_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspHi_selected");
}
void on_SuspMid_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspMid_selected");
}
void on_SuspLo_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspLo_selected");
}
void on_SuspFree_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspFree_selected");
}
void on_HighSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_HighSp_selected");
  SpSelected = 0;
  setMenu = Settings;
  menuSelected = true;
}
void on_MidSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_MidSp_selected");
  SpSelected = 1;
  setMenu = Settings;
  menuSelected = true;
}
void on_LowSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_LowSp_selected");
  SpSelected = 2;
  setMenu = Settings;
  menuSelected = true;
}
void on_SlowDownSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_SlowDownSp_selected");
  SpSelected = 3;
  setMenu = Settings;
  menuSelected = true;
}
void on_IgnDelaySp_selected(MenuItem* p_menu_item) {
  Serial.println("on_IgnDelay_selected");
  SpSelected = 4;
  setMenu = Settings;
  menuSelected = true;
}
void on_AlarmDelaySp_selected(MenuItem* p_menu_item) {
  Serial.println("on_AlarmDelay_selected");
  SpSelected = 5;
  setMenu = Settings;
  menuSelected = true;
}

void displayMenu() {
  lcd.clear();
  lcd.setCursor(0, 0);
  // Display the menu
  Menu const* cp_menu = ms.get_current_menu();
  lcd.print(cp_menu->get_selected()->get_name());
}

void incSp() {
  ++Setpoints[SpSelected];
  lcd.setCursor(0, 1);
  lcd.print(Setpoints[SpSelected]);
  lcd.setCursor(15, 1);
}

void decSp() {
  --Setpoints[SpSelected];
  lcd.setCursor(0, 1);
  lcd.print(Setpoints[SpSelected]);
  lcd.print(" ");
  lcd.setCursor(15, 1);
}

void serialHandler() {
  char inChar;
  if((inChar = Serial.read())>0) {

    switch (inChar) {
    case 'w': // Previus item
        if (!menuSelected) {
          ms.prev();
          displayMenu();
        }
        else {
          incSp();
        }
      break;
    case 's': // Next item
        if (!menuSelected) {
          ms.next();
          displayMenu();
        }
        else {
          decSp();
        }
      break;
    case 'a': // Back presed
      if(!menuSelected)
        ms.back();
        menuSelected = false;
        setMenu = Other;
        lcd.noBlink();
        displayMenu();
      break;
    case 'd': // Select presed
        if (!menuSelected) {
          ms.select();
          displayMenu();
          if (setMenu == Settings) {
            lcd.setCursor(0, 1);
            lcd.print(Setpoints[SpSelected]);
            lcd.setCursor(15, 1);
            lcd.blink();
          }
        }
        else {
          menuSelected = false;
          lcd.noBlink();
        }
      break;
    case 'e': // Cancel (clear menuSelected) For debug purpose
      menuSelected = false;
      break;
    } 
  }
}

void setup() {
  Serial.begin(9600);
  // set up the LCD's number of columns and rows:
  lcd.begin(16, 2);
  lcd.setCursor(0, 0);
//  lcd.setBacklight(RED);
  lcd.noBlink();
  lcd.print("Setup");

  // Menu setup
  mm.add_item(&mm_mi1, &on_SuspHi_selected);
  mm.add_item(&mm_mi2, &on_SuspMid_selected);
  mm.add_item(&mm_mi3, &on_SuspLo_selected);
  mm.add_item(&mm_mi4, &on_SuspFree_selected);

  mm.add_menu(&mu1);//Settings
  mu1.add_item(&mu1_mi1, &on_HighSp_selected);
  mu1.add_item(&mu1_mi2, &on_MidSp_selected);
  mu1.add_item(&mu1_mi3, &on_LowSp_selected);
  mu1.add_item(&mu1_mi4, &on_SlowDownSp_selected);
  mu1.add_item(&mu1_mi5, &on_IgnDelaySp_selected);
  mu1.add_item(&mu1_mi6, &on_AlarmDelaySp_selected);
  ms.set_root_menu(&mm);

  menuSelected = false;
  setMenu = Other;
  displayMenu();
}`

jonblack, 2016-04-04:
Let's see if I understand. You want to show the value behind a setting in the menu along with the setting. Is that correct? What happens if they select the menu item? Should the user also be able to change the value when on the same menu item?

recyclix, 2016-04-05:
Yes, I want to show the value associated with a menu item along with the item name when scrolling a particular menu.
In the code provided; when the item is selected the user can see the value and change it.
So if the user wants to check the value of each setpoint they have to select item/unselect item/ scroll menu...repeat...tedious and introduces the possibility of accidentally changing a setpoint 

If the name of menu currently selected (not item) and position of currently displayed item is known; a switch case can go in 'display menu' to provide any information required for that item, such as a value/description/instruction.
Once the user selects the item they are able to act on it.

Much as in Issue 20; when the menu is selected items are added dynamically.
What is available to indicate a particular menu (not item) is selected?

recyclix, 2016-04-05:
In this example I use variables (setMenu) and [menuItem] to select and change setpoints stored in an array through the menu system.
If; when the "Settings" menu was selected (setMenu) = Settings and [menuItem] = position of displayed menu item then there is no need for individual callbacks for each menu item; and potentially 100's of setpoints can be handled pretty easily.

It looks like the functions 'ms.get_current_menu_name()' and 'ms.get_cur_menu_item()' should be the answer but I get an error using them?

`
// include the library code:
#include <LiquidCrystal.h>
LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

//#include <Adafruit_RGBLCDShield.h>
//#include <utility/Adafruit_MCP23017.h>
// Initiate Display
//Adafruit_RGBLCDShield lcd = Adafruit_RGBLCDShield();
#define RED 0x1

#include <MenuSystem.h>
// Menu variables
MenuSystem ms;
Menu mm("");
MenuItem mm_mi1("HIGH ");
MenuItem mm_mi2("MID");
MenuItem mm_mi3("LOW");
MenuItem mm_mi4("FREE");
Menu mu1("SETTINGS>");
MenuItem mu1_mi1("HighSp");
MenuItem mu1_mi2("MidSp");
MenuItem mu1_mi3("LowSp");
MenuItem mu1_mi4("SlowDownSp");
MenuItem mu1_mi5("IgnDelaySp");
MenuItem mu1_mi6("AlarmDelaySp");

enum setMenuSelected_Type {
  Settings, Other
}; // this enum is in case we want to expand this example
setMenuSelected_Type setMenu;
boolean menuSelected = false;

int Setpoints[] {1000, 750, 500, 100, 80, 10};
int menuItem;

void loop() {
  serialHandler();
} // end loop

//Menu operations
void on_SuspHi_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspHi_selected");
}
void on_SuspMid_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspMid_selected");
}
void on_SuspLo_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspLo_selected");
}
void on_SuspFree_selected(MenuItem* p_menu_item) {
  Serial.println("on_SuspFree_selected");
}
void on_HighSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_HighSp_selected");
  menuItem = 0;
  setMenu = Settings;
  menuSelected = true;
}
void on_MidSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_MidSp_selected");
  menuItem = 1;
  setMenu = Settings;
  menuSelected = true;
}
void on_LowSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_LowSp_selected");
  menuItem = 2;
  setMenu = Settings;
  menuSelected = true;
}
void on_SlowDownSp_selected(MenuItem* p_menu_item) {
  Serial.println("on_SlowDownSp_selected");
  menuItem = 3;
  setMenu = Settings;
  menuSelected = true;
}
void on_IgnDelaySp_selected(MenuItem* p_menu_item) {
  Serial.println("on_IgnDelay_selected");
  menuItem = 4;
  setMenu = Settings;
  menuSelected = true;
}
void on_AlarmDelaySp_selected(MenuItem* p_menu_item) {
  Serial.println("on_AlarmDelay_selected");
  menuItem = 5;
  setMenu = Settings;
  menuSelected = true;
}

void spDisplay() {
  lcd.setCursor(0, 1);
  lcd.print(Setpoints[menuItem]);
  lcd.print(" ");
  lcd.setCursor(15, 1);
  lcd.blink();
//  Serial.print(Setpoints[menuItem]);
}

void displayMenu() {
  lcd.clear();
  lcd.setCursor(0, 0);
  // Display the menu
  Menu const* cp_menu = ms.get_current_menu();
  lcd.print(cp_menu->get_selected()->get_name());
//  Serial.print(cp_menu->get_selected()->get_name());

  switch (setMenu) {
    case Settings:
      spDisplay();
      break;
    case Other:
      //do something
      break;
    default:
      break;
  }
//Serial.println("displayMenu Called");
}

void serialHandler() {
  char inChar;
  if ((inChar = Serial.read()) > 0) {

    switch (inChar) {
      case 'w': // Previus item
        if (!menuSelected) {
          ms.prev();
          displayMenu();
        }
        else {
          switch (setMenu) {
            case Settings:
              ++Setpoints[menuItem];
              spDisplay();
              break;
            case Other:
              //do something
              break;
            default:
              break;
          }
        }
        break;
      case 's': // Next item
        if (!menuSelected) {
          ms.next();
          displayMenu();
         }
        else {
          switch (setMenu) {
            case Settings:
              --Setpoints[menuItem];
              spDisplay();
              break;
            case Other:
              //do something
              break;
            default:
              break;
          }
        }
        break;
      case 'a': // Back presed
        if (!menuSelected)
          ms.back();
        menuSelected = false;
        setMenu = Other;
        lcd.noBlink();
        displayMenu();
        break;
      case 'd': // Select presed
        if (!menuSelected) {
          ms.select();
          displayMenu();
        }
        else {
          menuSelected = false;
          lcd.noBlink();
        }
        break;
      case 'e': // Cancel (clear menuSelected) For debug purpose
        menuSelected = false;
        break;
    }
  }
}

void setup() {
  Serial.begin(9600);
  // set up the LCD's number of columns and rows:
  lcd.begin(16, 2);
  lcd.setCursor(0, 0);
  lcd.setBacklight(RED);
  lcd.noBlink();
  lcd.print("Setup");

  // Menu setup
  mm.add_item(&mm_mi1, &on_SuspHi_selected);
  mm.add_item(&mm_mi2, &on_SuspMid_selected);
  mm.add_item(&mm_mi3, &on_SuspLo_selected);
  mm.add_item(&mm_mi4, &on_SuspFree_selected);

  mm.add_menu(&mu1);//Settings Menu
  mu1.add_item(&mu1_mi1, &on_HighSp_selected);
  mu1.add_item(&mu1_mi2, &on_MidSp_selected);
  mu1.add_item(&mu1_mi3, &on_LowSp_selected);
  mu1.add_item(&mu1_mi4, &on_SlowDownSp_selected);
  mu1.add_item(&mu1_mi5, &on_IgnDelaySp_selected);
  mu1.add_item(&mu1_mi6, &on_AlarmDelaySp_selected);
  ms.set_root_menu(&mm);

  menuSelected = false;
  setMenu = Other;
  displayMenu();
}
`

jonblack, 2016-04-09:
Another user has created a pull request (#22)  which might also solve this issue for you. Maybe you can take a look?

recyclix, 2016-04-10:
Thanks Jon, this will be useful;
added a menu item that has a 'numeric modal' mode, in this mode it uses ```next()``` and ```prev()``` input to increase and decrease its numeric value. ```select()``` toggles the modal mode.

But doesn't solve how to display setpoint without selecting menu item.

eadf, 2016-04-10:
Each NumericMenuItem displays a computed/generated text on each get_name() call. So when scrolling down the menu you can see the name & value of your menu item. e.g. "Set point=14". 

Take a look at the `serial_nav.ino` example I just [updated](https://github.com/lacklustrlabs/arduino-menusystem).

recyclix, 2016-04-11:
Very nice; :+1: Flexible and saves a lot of code!
I love the limits & step size & display options. 

How can I use it with array or struct to save setpoints to eeprom or sd card so the machine remembers it's state?
 ```NumericMenuItem mm_mi5("Level 1 - Int Item 5 (Item)", menuName[0], -100, 100, 1, intMenuFormat);```
To load at startup and save when backing out of the menu.

eadf, 2016-04-11:
> How can I use it with array or struct to store 'value' on eeprom or sd card?
> so the machine remembers it's state?

Hm, I haven't considered any such functionality. But you can always iterate over all your NumericMenuItems, call get_value() on them and store that on eeprom.
Then when the device boots up again it can read the values from eeprom and use that in the   NumericMenuItems constructor.

But it would be nice if MenuSystem could store and create menus in a binary format.

jonblack, 2016-04-13:
I created an issue for serializing the menu system state so users can persist it. See #25.

jonblack, 2016-06-02:
Closing because the master branch contains #22 which solves this problem. Re-open if you feel this doesn't address your problem.

closed 2016-06-02 by jonblack