jonblack/arduino-menusystem #23: set-name

StyletX, 2016-04-12:
Hi! How to use the set-name function?
`#include <MenuSystem.h>

MenuSystem ms;
Menu mm("Выключен    ");

void setup() {
  ms.set_root_menu(&mm);
}

void loop() {
  test();
}

void test(){
  Menu const* cp_menu = ms.get_current_menu();
  MenuComponent const* cp_m_comp = cp_menu->get_selected(); //
  if (cp_m_comp->get_name() == "Выключен    ") cp_m_comp->set_name("Включен     ");
      else cp_m_comp->set_name("Выключен    ");
}
`

I receive an error:
Arduino\sketch_apr12a\sketch_apr12a.ino: In function 'void test()':

sketch_apr12a:17: error: passing 'const MenuComponent' as 'this' argument of 'void MenuComponent::set_name(const char*)' discards qualifiers [-fpermissive]

sketch_apr12a:18: error: passing 'const MenuComponent' as 'this' argument of 'void MenuComponent::set_name(const char*)' discards qualifiers [-fpermissive]

exit status 1
passing 'const MenuComponent' as 'this' argument of 'void MenuComponent::set_name(const char*)' discards qualifiers [-fpermissive]

jonblack, 2016-04-12:
`get_selected` returns a `MenuComponent const*`, which means that you can't change the name.

To be honest, I'm not sure why there is a `set_name` method. I'm not sure why I added it and can't see any reason why it would be useful. The expected use is to give a `MenuComponent` a name via its constructor and never change it, which is why a lot of functions return objects as `const`.

Why do you need to change the name?

StyletX, 2016-04-12:
Example
Menu->menu item 1->value
For example LED brightness:
Settings->Led bright->0...255 (changes)

jonblack, 2016-04-12:
That's not possible with this library (yet).

When I created the library I didn't view changing settings as part of the menu. So `LED brightness` is a menu item that when selected, takes you to a screen to change the brightness that isn't managed by this library.

It seems like this is a popular use-case, however, and thankfully @lacklustrlabs has made a pull-request (#22) to add support, so you're welcome to try that out. I'm not sure when it will become part of a new release.

StyletX, 2016-04-12:
Ok, thx!

closed 2016-04-12 by StyletX