jonblack/arduino-menusystem #24: get_selected() misnomer

eadf, 2016-04-12:
There seems to be a dual meaning of the term 'selected' in the library.
For example when we scroll down the list of items in a menu, calling`next()`on the MenuSystem, we get a new menu item from`get_selected()`.  But in this scenario the`select()`function has never been called. So how can the result of`get_selected()`change?

Maybe`get_current()`should be a more appropriate name? (just like the internal index `_cur_menu_component_num` indicates)
 
I understand that this is a breaking change, I'm just writing this down so that it won't be forgotten when you decide to make the next major release.

jonblack, 2016-04-13:
You're right that `selected` is the wrong term. I'm guessing (it was a while ago) that I was thinking in terms of the `MenuSystem::select()` method, although the index is tracked with `_cur_menu_component_num` which implies "current" and not "selected", so perhaps I just wasn't awake at the time :smile:

I don't mind adding the renamed version and introducing a warning for the next release.

jonblack, 2016-06-02:
Added `MenuSystem::get_current_component` and updated all the examples to call it. `get_selected` calls `get_current_component` for now and in a future release will be removed.

closed 2016-06-02 by jonblack