jonblack/arduino-menusystem #26: Add on_select parameter to MenuItem constructor

jonblack, 2016-04-21:
Currently the `on_select` callback is set using `set_selected_function` inside `Menu::add_item`. It makes more sense to add a parameter for the callback in the `MenuItem` constructor.

Deprecate the current `add_item` signature and add another one without the callback parameter.

Can `set_select_function` be removed after these changes? I think so, but perhaps there's a good use case for it.

eadf, 2016-04-21:
One use case where`set_select_function`would be needed is if a state machine is using the select function as the state. When the state is changed, the select function changes with it. Avoiding huge switch/case blocks could make code more readable. It could also make code more spaghetti like :)

I would have kept it, a one liner like that can't be many bytes.

jonblack, 2016-04-21:
I was thinking about dynamic `MenuItem`'s when I wrote that. I'm wondering if that can be supported in a better way than switching out the contents (name, callback, etc) of an already created instance. I mainly put that comment above to remind me to think about it.

jonblack, 2016-06-02:
This has been implemented in the master branch. Some of the examples have been updated to reflect the changes.

The old `add_item` and constructors have been marked as deprecated for future removal.

closed 2016-06-02 by jonblack