jonblack/arduino-menusystem #31: Menu select() callback missing

gzabusek, 2016-04-26:
Hello there,

in my case it would be useful to provide also a callback method for when a sub-Menu is selected. It would allow for clean code. Right now we have to check this manually whether a specific menu was selected. Can you please let me know if I just missed something? If not I can start working on the implementation of this today. Let me know if you would like that. Thanks for great lib!

jonblack, 2016-06-02:
A `Menu` is considered a container for `MenuComponent`'s. Any actions associated with your menu are assumed to be associated with a `MenuItem`. When you `select` a `Menu`, the only expected result is to show what's in that menu.

I'm interested to know why you need to know if a `Menu` has been selected?

gzabusek, 2016-06-02:
Hello there, its been a while :) But the reason was (i hope i remember precisely) that if one is using dynamic menus like in my case (ie - the submenu is only generated after the parent menu is selected - for instance when the menu is fetched from the network etc). For these reason (allocating / deallocating dynamic menus) it would be useful to have this.

jonblack, 2016-06-02:
That's a pretty good reasonable use case. I'll keep this open and take a look at it with #20.

jonblack, 2016-06-15:
I'm gonna close this and resolve it in issue #20. That may not be for the next release.

closed 2016-06-15 by jonblack