jonblack/arduino-menusystem #39: Edit a small typo and its running on ESP8266 Arduino

artkeller, 2016-07-10:
As the ESP8266 Arduino implementation is a bit more strict with types, in this 32bit environment the types **bool** and **boolean** are not identical.
However, the following edit in **MenuSystem.cpp** will help:

`void MenuSystem::select(boolean reset){...}`

BTW1: On ESP8266 Arduino you will NOT have the same SRAM memory pain anymore as on UNO, etc...
BTW2: ESP8266 Arduino is very comfortable in OTA projects - and part of a nice hardware ecosystem (WeMos D1, D1 mini, nodemcu, etc.)

jonblack, 2016-07-10:
Thanks. This has already been fixed in master. Everything should indeed be `bool`.

You're right about the other boards. Even the MEGA makes life a little easier. That said, the UNO board is so popular I want to try and support it as best I can.

closed 2016-07-10 by jonblack

artkeller, 2016-07-13:
Thanks for your quick response. 
Let me propose a toast about "resources": 

I'm an UNO fan as well and I like all the tiny ones like ATtiny25, 45, 85, etc. In my former days I worked  for some years in my real life with this "old languages" like COBOL68, FORTRAN4, ALGOL, LISP and all this stuff on Minis like PDP and Mainframes with MVS  - and at night on ASM basis with all the very early Micros like the 4004, 4040, 8008, 8080, the polular 6502, the forgotten 6800- and 68000-family  - at that time we were real incredibly crazy people - some called us "nerds". And I remember the bit-counting-business and the minimalistic approach very well. This vintage stuff in mind makes me happy to see nowadays someone like you will have a heart for the tiny ones! Thanks!

IoT is real and rising up every day... so we need the knowledge about small resource handling beyond all that ui-based framework oo-stuff. Embedded systems seem to be more sexy sometimes, more real  than all the nice "hollywood platforms".  Due to the scaling effect of IoT, which will bring massive data load into the networks,  I feel that it is crucial to handle resources sensitively now and in future again - like in earlier days. Did we all learn the lessions?