jonblack/arduino-menusystem #40: Examples compile error

etkArduino, 2016-07-25:
Compiling several examples on Arduino IDE 1.6.3 I get an error like this (serial_nav.ino):

>>>>>>
Arduino: 1.6.3 (Windows 8.1), Board:"Arduino Nano, ATmega328"
In file included from CustomNumericMenuItem.h:13:0,
                 from CustomNumericMenuItem.cpp:1:
D:\Documents\Arduino\sketchbook\libraries\MenuSystem/MenuSystem.h:180:11: error: expected nested-name-specifier before 'SelectFnPtr'
     using SelectFnPtr = void (*)(MenuItem* menu_item);
               ^
D:\Documents\Arduino\sketchbook\libraries\MenuSystem/MenuSystem.h:180:11: error: using-declaration for non-member at class scope
D:\Documents\Arduino\sketchbook\libraries\MenuSystem/MenuSystem.h:180:23: error: expected ';' before '=' token
     using SelectFnPtr = void (*)(MenuItem* menu_item);
                                 ^
D:\Documents\Arduino\sketchbook\libraries\MenuSystem/MenuSystem.h:180:23: error: expected unqualified-id before '=' token
D:\Documents\Arduino\sketchbook\libraries\MenuSystem/MenuSystem.h:188:32: error: 'SelectFnPtr' has not been declared
     MenuItem(const char* name, SelectFnPtr select_fn);
                                                            ^
<<<<<
and more ....

Am I missing something?

jonblack, 2016-07-26:
The library uses features from c++11 which require compiling with `-std=c++11` or `-std=gnuc++11`. This is the default in version 1.6.6 of the IDE, so if you upgrade it should work fine. Let me know if it doesn't.

etkArduino, 2016-07-27:
Jon,

I installed IDE version 1.6.10 and it now compiles.
Didn't realize that such a major change is just denoted by a minor update
version number ...
Thanks for your help!

I now have to sort out why my Arduino Nano is not recognized by my laptop,
i.e. the sketch won't upload.
But that's another (driver) issue.

Again: thanks!

Regards,

Eric

closed 2016-07-27 by etkArduino