jonblack/arduino-menusystem #42: Multilanguage menu

RobiGoBH, 2016-11-17:
Using the serial_nav example I tried to create multi language menu.
In the serial_nav.ino I defined a few variables like 
`char* l_eng_mu_mm_set = "SETTINGS";
char* l_hun_mu_mm_set = "BEALLITASOK";`

then I set the menu name to l_eng_mu_mm_set  and  I created a translate function, so if I change the language then instead of printing SETTINGS call the translate function and print out BEALLITASOK.
Something like:
Serial.print(translate(menu_item.get_name()));

The problem, I can't call in MyRenderer.cpp the translate function declared in serial_nav.ino
And for changing the language I think it will be good the colour changer NumericMenuItem from the same example, but can you tell me how to get the value of that menu item to use in translate function?

Can you help me how to call that or how to deal with this situation?

Thank you.

jonblack, 2016-11-19:
You could add your `translate` function to your renderer and call that to get the text that you want to display before calling `Serial.print`.

You could take this a step further and create a separate module for loading text. Look at projects like [`GNU gettext`](https://www.gnu.org/software/gettext/gettext.html) for inspiration. This module you create would then be a member of your renderer which you'd call the get the menu text your interested in before displaying it.

RobiGoBH, 2016-11-19:
Thank you Jon,  the gettext looks like a good inspiration source. 
Can you tell me please, how I can get the NumericMenuItem value? (I would like to use one to store the selected language) I need to iterate through all items or is it possible to refer somehow direct to that menuitem?

jonblack, 2016-11-20:
You can use `NumericMenuItem::get_value` or `NumericMenuItem::get_string_value` to get the current value of the menu item. This is most often used inside the `render` function to display said value on whatever display you are using.

RobiGoBH, 2016-11-20:
Thank you, I will use that temporary, and I'm waiting so much the serializable version to be released to read and write the menu content to SD card much easier.

closed 2016-11-21 by jonblack