jonblack/arduino-menusystem #46: NumericMenuItem Doesn't Increment

dgbalharrie, 2017-01-31:
NumericMenuItem next() isn't called.  if (_p_curr_menu->_p_current_component->has_focus()) test fails, if 
return _p_curr_menu->_p_current_component->next(loop); is called first in bool MenuSystem::next(bool loop) it does increment.

dgbalharrie, 2017-01-31:
Further investigation. It does work, but I have to select twice the first time to activate the increment. That is Menu* NumericMenuItem::select() only gets called the second time ms.select() is selected. My menu structure is menu->menu->numericMenuItem.

dgbalharrie, 2017-02-07:
Well, I can only assume this is working as designed. You need to select to display the menuitem, then select again to edit the item? I have created two new menuitem classes. NumericMenuItemInt to display and inc/dec an int item. I also created an ArrayMenuItem, which creates one item for multiple strings e.g. "Yes", "No" etc. With both of these classes I  can also test against a preset value to display an asterix (or other character) , to show the value that  is set in the host program.

jonblack, 2017-02-07:
Sorry for the late reply. Your last comment is correct: you need to first select it to display the menu item and then again to enter edit mode. The `serial_nav` example shows this if you need to get a feel for it working in practice.

dgbalharrie, 2017-02-07:
Thank you for the reply. I think what puzzled me was that MenuItems did not need this. As noted I am now using an object called ArrayMenuItem for none numerical items.
For example:-
Menu baudMenu("Serial->Baud");
const char *rates[] = { "2400", "4800", "9600", "19200", "38400" };
ArrayMenuItem baudrate_mi("Baudrate = ", rates, 5, 2, on_item4_selected);
The value 2 is the index value used to determine the item in the array that will display an asterix in the display method using curItemMatch()
	virtual void render_array_menu_item(ArrayMenuItem const& menu_item) const
	{
		lcd.print(menu_item.get_name());
		//		lcd.print(" ");
		lcd.print(menu_item.get_selItem_string());
		if (menu_item.curItemMatch() == true) lcd.print("*");
	}
The same is true for NumericMenuItemInt.
Not sure if the custommenu would have provided the same, but playing with the above was good to help understand some more of the library!

jonblack, 2017-02-08:
Great to hear your making progress. Can I close this issue now?

dgbalharrie, 2017-02-08:
Yes, close the issue.

 

This email has been scanned by BullGuard antivirus protection.
For more info visit www.bullguard.com <http://www.bullguard.com/tracking.aspx?affiliate=bullguard&buyaffiliate=smtp&url=/>

closed 2017-02-08 by jonblack