jonblack/arduino-menusystem #47: serial_nav does not compile on Arduino 1.8.0

milkpirate, 2017-02-02:
Hey,

I wanted to try out your menu library (already looking for one which isnt over complicated, works and isnt 5 years old) so I tried to compile the `serial_nav` example but it gives me:

```
Sketch wird kompiliert...
"C:\Program Files (x86)\Electronics\arduino-1.8.0\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MINI -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Electronics\arduino-1.8.0\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Electronics\arduino-1.8.0\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Users\XXX\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem" "C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.cpp" -o "C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.cpp.o"
In file included from C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.h:14:0,

                 from C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.cpp:1:

MyRenderer.h:19: error: expected class-name before '{' token

 {

 ^

MyRenderer.h:23: error: 'BackMenuItem' has not been declared

     virtual void render_back_menu_item(BackMenuItem const& menu_item) const;

                                        ^

MyRenderer.h:24: error: 'NumericMenuItem' has not been declared

     virtual void render_numeric_menu_item(NumericMenuItem const& menu_item) const;

                                           ^

In file included from C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.cpp:1:0:

CustomNumericMenuItem.h:19: error: expected class-name before '{' token

 {

 ^

CustomNumericMenuItem.h:33: error: 'FormatValueFnPtr' has not been declared

                           FormatValueFnPtr on_format_value=nullptr);

                           ^

CustomNumericMenuItem.h:37: error: 'MenuComponentRenderer' has not been declared

     virtual void render(MenuComponentRenderer const& renderer) const;

                         ^

CustomNumericMenuItem.h:33: error: could not convert 'nullptr' from 'std::nullptr_t' to 'int'

                           FormatValueFnPtr on_format_value=nullptr);

                                                            ^

CustomNumericMenuItem.cpp:5: error: 'FormatValueFnPtr' has not been declared

         float maxValue, float increment, FormatValueFnPtr on_format_value)

                                          ^

C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.cpp: In constructor 'CustomNumericMenuItem::CustomNumericMenuItem(uint8_t, const char*, float, float, float, float, int)':

CustomNumericMenuItem.cpp:6: error: class 'CustomNumericMenuItem' does not have any field named 'NumericMenuItem'

 : NumericMenuItem(name, nullptr, value, minValue, maxValue, increment,

   ^

C:\Users\XXX\AppData\Local\Temp\arduino_build_442257\sketch\CustomNumericMenuItem.cpp: At global scope:

CustomNumericMenuItem.cpp:17: error: variable or field 'render' declared void

 void CustomNumericMenuItem::render(MenuComponentRenderer const& renderer) const

                                                          ^

CustomNumericMenuItem.cpp:17: error: 'MenuComponentRenderer' was not declared in this scope

 void CustomNumericMenuItem::render(MenuComponentRenderer const& renderer) const

                                    ^

Mehrere Bibliotheken wurden für "MenuSystem.h" gefunden
 Benutzt: C:\Users\XXX\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem
 Nicht benutzt: C:\Users\XXX\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem-master
Bibliothek arduino-menusystem in Version 2.1.1 im Ordner: C:\Users\XXX\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem  wird verwendet
exit status 1
expected class-name before '{' token
```

Am I missing obvious?

jonblack, 2017-02-03:
Are you using the Arduino IDE to compile your code or something else? If you are using the IDE, which version are you using?

jonblack, 2017-02-03:
Using [Arduino-Makefile](https://github.com/sudar/Arduino-Makefile) and the `Makefile` included with the example, I'm able to compile without any errors using version 1.8.1 of the Arduino IDE.

Have you made changes to the serial_nav sketch?

During compilation it comes across the `NumericMenuItem` base class of `CustomNumericMenuItem` and doesn't know what it is. That should have been included via `MenuSystem.h` but it looks as though g++ on Windows is processing it differently (just a guess, I have no idea).

Try adding the following to `CustomNumericMenuItem.h`:

```
#ifndef _CUSTOMNUMERICMENUITEM_H
#define _CUSTOMNUMERICMENUITEM_H

#include <MenuSystem.h>
#include "MyRenderer.h"

class MyRenderer;
class NumericMenuItem;   // <--- this is new
```

Sadly I can't test this on Windows because I don't have Windows on any machines.

milkpirate, 2017-02-07:
No havent changed anything. Tried your suggestion. Kinda works. Now it's having trouble with `MyRenderer.h`:

```
Sketch wird kompiliert...
"C:\Program Files (x86)\Electronics\arduino-1.8.0\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10800 -DARDUINO_AVR_MINI -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Electronics\arduino-1.8.0\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Electronics\arduino-1.8.0\hardware\arduino\avr\variants\eightanaloginputs" "-IC:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem" "C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.cpp" -o "C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.cpp.o"
In file included from C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.h:14:0,

                 from C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.cpp:1:

MyRenderer.h:19: error: expected class-name before '{' token

 {

 ^

MyRenderer.h:23: error: 'BackMenuItem' has not been declared

     virtual void render_back_menu_item(BackMenuItem const& menu_item) const;

                                        ^

MyRenderer.h:24: error: 'NumericMenuItem' has not been declared

     virtual void render_numeric_menu_item(NumericMenuItem const& menu_item) const;

                                           ^

In file included from C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.cpp:1:0:

CustomNumericMenuItem.h:19: error: invalid use of incomplete type 'class NumericMenuItem'

 class CustomNumericMenuItem : public NumericMenuItem

                                      ^

CustomNumericMenuItem.h:17: error: forward declaration of 'class NumericMenuItem'

 class NumericMenuItem; 

       ^

CustomNumericMenuItem.h:34: error: 'FormatValueFnPtr' has not been declared

                           FormatValueFnPtr on_format_value=nullptr);

                           ^

CustomNumericMenuItem.h:38: error: 'MenuComponentRenderer' has not been declared

     virtual void render(MenuComponentRenderer const& renderer) const;

                         ^

CustomNumericMenuItem.h:34: error: could not convert 'nullptr' from 'std::nullptr_t' to 'int'

                           FormatValueFnPtr on_format_value=nullptr);

                                                            ^

CustomNumericMenuItem.cpp:5: error: 'FormatValueFnPtr' has not been declared

         float maxValue, float increment, FormatValueFnPtr on_format_value)

                                          ^

C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.cpp: In constructor 'CustomNumericMenuItem::CustomNumericMenuItem(uint8_t, const char*, float, float, float, float, int)':

CustomNumericMenuItem.cpp:6: error: type 'NumericMenuItem' is not a direct base of 'CustomNumericMenuItem'

 : NumericMenuItem(name, nullptr, value, minValue, maxValue, increment,

   ^

C:\Users\Lenny\AppData\Local\Temp\arduino_build_991837\sketch\CustomNumericMenuItem.cpp: At global scope:

CustomNumericMenuItem.cpp:17: error: variable or field 'render' declared void

 void CustomNumericMenuItem::render(MenuComponentRenderer const& renderer) const

                                                          ^

CustomNumericMenuItem.cpp:17: error: 'MenuComponentRenderer' was not declared in this scope

 void CustomNumericMenuItem::render(MenuComponentRenderer const& renderer) const

                                    ^

Mehrere Bibliotheken wurden für "MenuSystem.h" gefunden
 Benutzt: C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem
 Nicht benutzt: C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem-master
Bibliothek arduino-menusystem in Version 2.1.1 im Ordner: C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem  wird verwendet
exit status 1
expected class-name before '{' token
```

Tried (similar to the first fix):

```
#ifndef _MY_RENDERER_H
#define _MY_RENDERER_H

#include <MenuSystem.h>
#include "CustomNumericMenuItem.h"

class CustomNumericMenuItem;
class MenuComponentRenderer;       <-- added

class MyRenderer : public MenuComponentRenderer
```

But the it throws:

```
 Nicht benutzt: C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem-master
Bibliothek arduino-menusystem in Version 2.1.1 im Ordner: C:\Users\Lenny\Documents\Elektronik\Arduino\Scratch\libraries\arduino-menusystem  wird verwendet
exit status 1
invalid use of incomplete type 'class MenuComponentRenderer'
```

jonblack, 2017-02-07:
Because `MyRenderer` inherits from `MenuComponentRenderer` you need the full definition in `MyRenderer.h`, so you'd need to add the following to `MyRenderer.h`:

```
#include <MenuSystem.h>
#include <MenuComponentRenderer.h>  // <-- Add this line
#include "CustomNumericMenuItem.h"
```

This one is surprising because the error makes sense, the include should be there, so I have no idea why that works fine for me on Linux with the Makefile.

I think all of these errors are going to be resolved by either adding a forward declaration or including the appropriate header.

I will keep this issue open until I can reproduce it, which will probably mean getting a copy of Windows, but I'm quite busy at the moment so it won't happen soon.

If you keep getting stuck with the errors feel free to add a comment here and I'll help out.

milkpirate, 2017-02-08:
So tried these modification but the IDE doesnt find `MenuComponentRenderer.h`. It is not included in your repo, where do I find it? Is it a stock file of the IDE? What library do I have to install?

jonblack, 2017-02-08:
Ah my mistake I thought `MenuComponentRenderer` was in a separate file. It's actually in `MenuSystem.h`, so that's really odd that it doesn't work without that addition.

I'm not sure what to suggest now. When I get a chance I'll test it on Windows, and in the Arduino IDE to be sure.

milkpirate, 2017-02-08:
Ok, now it compiled (without `#include <MenuComponentRenderer.h>`):

`MyRenderer.h`:
```
#ifndef _MY_RENDERER_H
#define _MY_RENDERER_H

#include <MenuSystem.h>
#include "CustomNumericMenuItem.h"

class CustomNumericMenuItem;
class MenuComponentRenderer;

class MyRenderer : public MenuComponentRenderer
...
```
`CustomNumericMenuItem.h`:
```
#ifndef _CUSTOMNUMERICMENUITEM_H
#define _CUSTOMNUMERICMENUITEM_H

#include <MenuSystem.h>
#include "MyRenderer.h"

class MyRenderer;
class NumericMenuItem;

class CustomNumericMenuItem : public NumericMenuItem
...
```

jonblack, 2017-08-05:
I take it this is resolved now. Re-open if I'm mistaken.

closed 2017-08-05 by jonblack