jonblack/arduino-menusystem #48: Newbie! using data from NumericMenuItem

GotoNext, 2017-02-12:
Hello, newbie question.

I am using your menu system with great success. 
Now I am trying to to use te value in the `NumericMenuItem` in my own program, but because I am a newbie I would request some help.
I have manged to get the value of
 `CustomNumericMenuItem mu1_mi3(12, "Level 2 - Cust Item 3 (Item)", 80, 65, 121, 3, format_int);` 
but 1. not sure this is the right way 
2. how to handle the other menu items that use the 'format_int'
I hope my questions makes sense? if not please let me know and I will try to re-phrase it.
'
There my code :
```
`
#include <MenuSystem.h>
#include "CustomNumericMenuItem.h"
#include "MyRenderer.h"

int ledState = LOW;
String temp[10];
// forward declarations
const String format_float(const float value);
const String format_int(const float value);
const String format_color(const float value);   //color options !!!
const String stoomaan(const float value);  //to switch on and off the steamgenerator should be true or false  !!!

void on_item1_selected(MenuItem* p_menu_item);
void on_item2_selected(MenuItem* p_menu_item);
void on_item3_selected(MenuItem* p_menu_item);
void on_back_item_selected(MenuItem* p_menu_item);

// Menu variables

MyRenderer my_renderer;
MenuSystem ms(my_renderer);

Menu mm("Main Menu"); // this is possibly not needed ???
 
      NumericMenuItem mm_mi1("1. Stoomgenerator ", nullptr, 2, 0, 1, 1, stoomaan);// startwaarde,minimaal,maximaal,increment
      NumericMenuItem mm_mi2("2. Lamp aan/uit   ", nullptr, 2, 0, 1, 1, stoomaan);// startwaarde,minimaal,maximaal,increment

      Menu mu1("3. Settings> ");  // menu 2 menu item
      BackMenuItem mu1_mi1("Level 1 - Back (Item)", &on_back_item_selected, &ms); //menu 2 item 1

      MenuItem mu1_mi2("3.Settings in sub >", &on_item1_selected); //menu 2 item 2
      NumericMenuItem mu1_mi3("   3.1. tijds instelling", nullptr, 45, 0, 120, 1, format_int);  //startwaarde,minimaal,maximaal,increment 
      CustomNumericMenuItem mu1_mi4(12, "   3.2 temperatuurs instelling", 43, 20, 70, 1, format_int); // startwaarde,minimaal,maximaal,increment
      NumericMenuItem mm_mi4("5. - Float Item 4 (Item)", nullptr, 0.5, 0.0, 1.0, 0.1, format_float);
      NumericMenuItem mm_mi5("6. - Int Item 5 (Item)", nullptr, 50, -100, 100, 1, format_int);
    Menu mu2("4. level menu 2 ??");
  BackMenuItem mu2_mi1("Level 2 - Back (Item)", &on_back_item_selected, &ms);

MenuItem mu2_mi2("Level 2 - Item 1 (Item)", &on_item3_selected);
NumericMenuItem mu2_mi3("Level 2 - Txt Item 2 (Item)", nullptr, 0, 0, 3, 1, format_color);  // "??","??",aantal opties,"??"

/* original menu setup is below:   ECL
        
        MenuItem mm_mi1("Level 1 - Item 1 (Item)", &on_item1_selected);
        MenuItem mm_mi2("Level 1 - Item 2 (Item)", &on_item2_selected);
        Menu mu1("Level 1 - Item 3 (Menu)");
        BackMenuItem mu1_mi0("Level 2 - Back (Item)", &on_back_item_selected, &ms);
     
        MenuItem mu1_mi1("Level 2 - Item 1 (Item)", &on_item3_selected);
        NumericMenuItem mu1_mi2("Level 2 - Txt Item 2 (Item)", nullptr, 0, 0, 2, 1, format_color);
        CustomNumericMenuItem mu1_mi3(12, "Level 2 - Cust Item 3 (Item)", 80, 65, 121, 3, format_int);
        
        NumericMenuItem mm_mi4("Level 1 - Float Item 4 (Item)", nullptr, 0.5, 0.0, 1.0, 0.1, format_float);
        NumericMenuItem mm_mi5("Level 1 - Int Item 5 (Item)", nullptr, 50, -100, 100, 1, format_int);

and orginal taken from advance.ino sample from Jon Black

// Menu variables
MenuSystem ms;
Menu mm("Main Menu");
MenuItem mm_mi1("Read Analog 0 A0");
MenuItem mm_mi2("Blink led");
Menu mu1("Date & Time");
MenuItem mu1_mi1("Show current");
MenuItem mu1_mi2("Set");
MenuItem mu1_mi3("Go back");

*/

// Menu callback function

// writes the (int) value of a float into a char buffer.
const String format_int(const float value)
{
   Serial.println ("hier wordt de format_int aangepast ?:");
   Serial.println (value );
   temp[1]=( String((int) value));                         //////////  ECL copy menu value in my own integer    Serial.println (temp[1]);
   Serial.println(ms.get_current_menu()->get_name());      //////////  how to get value of other format_int ?
   return String((int) value);
   }

// writes the value of a float into a char buffer.
const String format_float(const float value)
{
  return String(value);
}

// writes the value of a float into a char buffer as settings.     ECL stoomaan toegevoegd
const String stoomaan(const float value)
{
  String buffer;

  switch ((int) value)
  {
    case 0:
      buffer += "uit";
      break;
    case 1:
      buffer += "aan";
      break;

    default:
      buffer += "uit";
  }

  return buffer;
}

// writes the value of a float into a char buffer as predefined colors.
const String format_color(const float value)
{
  String buffer;

  switch ((int) value)
  {
    case 0:
      buffer += "Red";
      break;
    case 1:
      buffer += "Green";
      break;
    case 2:
      buffer += "Blue";
      break;
    case 3:
      buffer += "Emiel";
      break;
    default:
      buffer += "undef";
  }

  return buffer;
}

// In this example all menu items use the same callback. //ECL shows only menu with subs in it???

void on_item1_selected(MenuItem* p_menu_item)
{
  Serial.println("Item1 Selected");

ledState = !ledState;
  //igitalWrite(led,ledState);
  //lcd.setCursor(0,1);
  Serial.print("Led state: ");
  if(ledState)
    Serial.print("ON");
  else
    Serial.print("OFF");
  //delay(SELECTED_DISPLAY_DELAY);
}
  
 
void on_item2_selected(MenuItem* p_menu_item)
{
  Serial.println("Item2 Selected");
}

void on_item3_selected(MenuItem* p_menu_item)
{
  Serial.println("Item3 Selected");
}

void on_back_item_selected(MenuItem* p_menu_item)
{
  Serial.println("Back item Selected");
}

void display_help() {
  Serial.println("***************");
  Serial.println("w: go to previus item (up)");
  Serial.println("s: go to next item (down)");
  Serial.println("a: go back (right)");
  Serial.println("d: select \"selected\" item");
  Serial.println("?: print this help");
  Serial.println("h: print this help");
  Serial.println("***************");
}

void serial_handler()
{
  char inChar;
  if ((inChar = Serial.read()) > 0)
  {
    switch (inChar)
    {
      case 'w': // Previus item
        ms.prev();
        ms.display();
        Serial.println("");
        break;
      case 's': // Next item
        ms.next();
        ms.display();
        Serial.println("");
        break;
      case 'a': // Back presed
        ms.back();
        ms.display();
        Serial.println("");
        break;
      case 'd': // Select presed
        ms.select();
        ms.display();
        Serial.println("");
        break;
      case '?':
      case 'h': // Display help
        ms.display();
        Serial.println("");
        break;
      default:
        break;
    }
  }
}

// Standard arduino functions

void setup()
{
  Serial.begin(9600);

  ms.get_root_menu().add_item(&mm_mi1);
  ms.get_root_menu().add_item(&mm_mi2);
  
  ms.get_root_menu().add_menu(&mu1);
  mu1.add_item(&mu1_mi1);
  //   ms.get_root_menu().add_item(&mm_mi2);
  //ms.get_root_menu().add_menu(&mu2);
  //mu1.add_item(&mu1_mi1);
  mu1.add_item(&mu1_mi2);
  mu1.add_item(&mu1_mi3);
  mu1.add_item(&mu1_mi4);
  //added by emiel
  ms.get_root_menu().add_menu(&mu2);
  mu2.add_item(&mu2_mi1);
  mu2.add_item(&mu2_mi2);
  mu2.add_item(&mu2_mi3);
  ms.get_root_menu().add_item(&mm_mi4);
  ms.get_root_menu().add_item(&mm_mi5);

  
  display_help();
  ms.display();
  Serial.println("");
  // stoomgenerator aan/uit functie als geselecteerd beginnen ECL
  ms.select();
  ms.display();
  Serial.println("");

}

void loop()
{
  serial_handler();`
```

jonblack, 2017-02-14:
I'm sorry I don't understand what your problem is. Can you post the line of code that's troubling you, telling me what happens vs what you expect to happen, then maybe I can help.

GotoNext, 2017-02-14:
Sorry I hope this explains better: 
In menu:

`CustomNumericMenuItem mu1_mi4(12, "   3.2 temperature  setting ", 43, 20, 70, 1, format_int); 
            NumericMenuItem mm_mi5("test  Int Item 5 (Item)", nullptr, 50, -100, 100, 1, format_int);`

I use the first 'format_int' like this :  

'// writes the (int) value of a float into a char buffer.
const String format_int(const float value) {
   temp[1]=( String((int) value));                                      //////////  ECL copy menu value in my own integer       

   return String((int) value);
   }
'
So I managed to use 'temp[1]' value but would like to have ' temp[2]' to contain 'mm_mi5' value. 
Now how do I get the value of 'mm_mi5' ???

In other words I need to have various values set in the menu, what is the best way to use these values in my code ?

jonblack, 2017-02-14:
The formatter used in the `NumericMenuItem` is for display purposes only. You want to use `float get_value() const;` on the `NumericMenuItem` to get its value, which you can use to update your settings.

Does that answer your question?

GotoNext, 2017-02-18:
I believe this might be the solution, only problem is that this goes beyond my abilities :(
Could you give me a example in on of the samples, from there I will be able to understand and implement. 
Just 1 line to get the value in a string (or other) and one line to print the string. 
(I hope :))

stathej, 2017-03-23:
Not tried it yet myself but give

temp[1]= mm_mi5.getvalue()

a try which I think should return the valve for mm_mi5, adjust for each menu level you want the value from and just block call to assign to your variables after you have done in the menu.  The menu will show the values as you change them just assign them when you have done.  You would need to format the value to suit as it will return the type used by the menu.

Or just declare unique calls at the end of each of the numeric menu items so that the values are saved as you change them, i.e change format_int to format_int1 and add code to handle format_int1 and repeat for each item.

jonblack, 2017-08-05:
My original comment still stands: the formatter callback is for telling the renderer how to display the value, so it shouldn't be used for that purpose.

If you really want to go down that road, you'd need a separate callback for each `NumericMenuItem`.

I think this is resolved now. Re-open if you disagree.

closed 2017-08-05 by jonblack