jonblack/arduino-menusystem #49: getting NumericMenuItem value in callback

dgbalharrie, 2017-02-15:
The menu is working nicely, but I can't seem to get the value that has been edited in NumericMenuItem in the callback. get_value() isn't available. If I change the parameter in the callback from MenuItem to NumericMenuItem it actually works, but I don't think this is correct.
Looking at the forums, I see others asking the same question but no answers.

jonblack, 2017-02-16:
You'll need to cast the value passed in to the callback:

```
void on_item_selected(MenuItem* p_menu_item)
{
    NumericMenuItem* p_nmi = dynamic_cast<NumericMenuItem*>(p_menu_item);
    Serial.println(p_nmi->get_value());
}
```

Make sure that the callback really is for a `NumericMenuItem`, otherwise you'll get a runtime error.

dgbalharrie, 2017-02-16:
Thank you for the information. However I get a compile error: 'dynamic_cast' not permitted with -fno-rtti
I have searched, but I can't find how you can change this option.

jonblack, 2017-02-16:
Ah sorry, there's no runtime-type information on the Arduino, so you can't use `dynamic_cast`. You'll have to do a c-style cast instead:

```
NumericMenuItem* p_nmi = (NumericMenuItem*) p_menu_item;
```

dgbalharrie, 2017-02-16:
That compiles ok. Thank you. I also realised that I know the objects in the callback and can access the objects directly which works as well as they are global on the arduino.
Thank you - you can close the issue.

closed 2017-02-18 by jonblack