jonblack/arduino-menusystem #51: Arduino Due not compiling

muluyilmaz, 2017-03-24:
This is a great library. Thanks John for this. I tried it with mega 2560 works perfect. but Could not compile with due board.

jonblack, 2017-03-24:
Can you paste the compilation error output here, please?

muluyilmaz, 2017-03-24:
Here are the errors ...  undefined reference to MenuSystem  
_____________________________________________________________________________________________
class MyRenderer : public MenuComponentRenderer
{
	
public:
    virtual void render(Menu const& menu) const
    {
		  lcd.clear();
		  
		
		  String theTitle = menu.get_name();
		  unsigned int TitleLenght = theTitle.length();
		  TitleLenght = ((round((40-TitleLenght)/2))-1);
		
		  lcd.setCursor(TitleLenght,0);
		  lcd.print(menu.get_name());
		  lcd.setCursor(0,0);
		  lcd.print(F(">>"));
		  lcd.setCursor(38,0);
		  lcd.print(F("<<"));
			
		 
		  lcd.setCursor(0,1);
		  menu.get_current_component()->render(*this);
		  
      	  lcd.setCursor(0,3);
      	  lcd.print(F(" ESC          ENT"));
		 
		
	    }

    virtual void render_menu_item(MenuItem const& menu_item) const
    {
	        lcd.print(menu_item.get_name());
    }

    virtual void render_back_menu_item(BackMenuItem const& menu_item) const
    {		
        lcd.print(menu_item.get_name());
	}

    virtual void render_numeric_menu_item(NumericMenuItem const& menu_item) const
    {
		
        lcd.print(menu_item.get_name());
	
		  
    }

    virtual void render_menu(Menu const& menu) const
    {
		  lcd.clear();
		  lcd.setCursor(0,0);
		  lcd.print(F("       MENU     "));
		  lcd.setCursor(0,1);
          lcd.print(menu.get_name());
		 
	
		  lcd.setCursor(0,3);
		  lcd.print(F("ESC                        ENT"));
		
    }

};
MyRenderer my_renderer;
MenuSystem ms(my_renderer);

___________________________________________________________________________________________________
DUeNEW.cpp.o*: In function MyRenderer::render_menu_item(MenuItem const&) const
DUeNEW.ino:473: undefined reference to MenuComponent  get_name() const
DUeNEW.cpp.o*: In function MyRenderer::render_back_menu_item(BackMenuItem const&) const
DUeNEW.ino:480: undefined reference to MenuComponent  get_name() const
DUeNEW.cpp.o*: In function MyRenderer::render_numeric_menu_item(NumericMenuItem const&) const
DUeNEW.ino:486: undefined reference to MenuComponent  get_name() const
DUeNEW.cpp.o*: In function MyRenderer::render_menu(Menu const&) const
DUeNEW.ino:499: undefined reference to MenuComponent  get_name() const
DUeNEW.cpp.o*: In function MyRenderer::render(Menu const&) const
DUeNEW.ino:448: undefined reference to MenuComponent  get_name() const
DUeNEW.cpp.o*: xxx...\AppData\Local\Temp\VMBuilds\DUeNEW\arduino_due_x_dbg\Debug\DUeNEW.cpp:547: more undefined references to MenuComponent::get_name() const follow
DUeNEW.cpp.o*: In function MyRenderer::render(Menu const&) const
DUeNEW.ino:461: undefined reference to Menu  get_current_component() const
DUeNEW.cpp.o*: In function SetMyMenu()
DUeNEW.ino:1458: undefined reference to MenuSystem  get_root_menu() const
DUeNEW.ino:1458: undefined reference to Menu  add_menu(Menu*)
DUeNEW.ino:1459: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1460: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1461: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1462: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1463: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.cpp.o*: xx....\AppData\Local\Temp\VMBuilds\DUeNEW\arduino_due_x_dbg\Debug\DUeNEW.cpp:1558: more undefined references to Menu::add_item(MenuItem*) follow
DUeNEW.cpp.o*: In function SetMyMenu()
DUeNEW.ino:1470: undefined reference to MenuSystem  get_root_menu() const
DUeNEW.ino:1470: undefined reference to Menu  add_menu(Menu*)
DUeNEW.ino:1471: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1472: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1473: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1474: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1475: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.cpp.o*: xxx...1\AppData\Local\Temp\VMBuilds\DUeNEW\arduino_due_x_dbg\Debug\DUeNEW.cpp:1570: more undefined references to Menu::add_item(MenuItem*) follow
DUeNEW.cpp.o*: In function SetMyMenu()
DUeNEW.ino:1478: undefined reference to MenuSystem  get_root_menu() const
DUeNEW.ino:1478: undefined reference to Menu  add_menu(Menu*)
DUeNEW.ino:1479: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1480: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1481: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1482: undefined reference to Menu  add_item(MenuItem*)
DUeNEW.ino:1483: undefined reference to Menu  add_item(MenuItem*)

jonblack, 2017-03-24:
Are you using the Arduino IDE? Have you included the library in the sketch?

muluyilmaz, 2017-03-25:
I am using Atmel studio 7 with vmicro which integrates arduino ide to Atmel studio. Yes I included library. But library is in arduino library directory . Should I copy it to an other directory for DUE. PS : tried it with arduino ide same errors when trying to compile for due but no error for uno and mega.

jonblack, 2017-03-28:
Is your entire sketch in a single source file or have your split it up?

muluyilmaz, 2017-03-28:
I solved the problem ... 

  I copy your library in to
 
**C:\Users\XXX\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.9\libraries**

and it compiled without errors. I  set up my breadboard with Keypad , and 4x40 lcd and try the Perfect and usefull MenuSystem .  No problems works as a charm ... 

Thank you again ..

closed 2017-03-28 by muluyilmaz