jonblack/arduino-menusystem #52: Add a setSelected and getSelected option

dpeddi, 2017-05-06:
Hi,

I'm here to request you for an enachement...

I'm trying to customize and improve following program to adapt to my usage: https://github.com/souliss-smart-thermostat-WiFi/souliss-smart-thermostat-WiFi/commit/1a2cb553af2ac84882bf909060656d55c252690d.

That thermostat make use of your library (an old release). They home made the menu title,  
and they used a "*" to differentiate the option currently in use... but it doesn't work as expected. So I've temporary added a big "if" to fix that issue. 

I've already tried to adapt that program to use your latest version, but to remove that if it would be nice to get a "setSelected" and "getSelected" option to store the last selected entry directly into the menu structure.

Do you think that such option may be useful to you and/or to other guys as well?

Regards

jonblack, 2017-05-09:
If I understand correctly, you're updating the program to use a new version of this library. The `is_selected` function is used to add a `*` to the output to indicate which menu item is currently selected. This is difficult because there's no `getSelected` option to test against, so you've got this big if block.

MenuSystem now uses a concept of a renderer, with a rendering method for each type of menu component to render. The render method has access to the component it's rendering and each component has an `is_current` method, which returns `true` if it's the currently selected component. The [serial_nav example here](https://github.com/jonblack/arduino-menusystem/blob/master/examples/serial_nav/MyRenderer.cpp#L13) does exactly this.

So I don't think `setSelected` or `getSelected` methods are necessary. What do you think?

jonblack, 2017-08-10:
No response so closing. If this is still an issue, feel free to re-open.

closed 2017-08-10 by jonblack