jonblack/arduino-menusystem #53: Arduinio DUE / ms.back & ms.select .

muluyilmaz, 2017-06-14:
Hi ; 

There is some error at ms.back and ms. select at DUE . 

I calll ms.back with keypad escape key. if I am at root menu and call ms.back code stops and freeze also same at ms. select if the slected item subroutine is empty . after both ms.back and ms.select I call ms.display. 

I checked your code ms.back section and ms.select section . 

it seems you took care of it  by  if current menu is  root menu return false . 

Can not understand why my DUE is freezing. Same code at Mega works perfect. 

Any suggestion please .. 

Thanks.

jonblack, 2017-06-20:
Thanks for reporting the issue. Without a code sample it's hard for me to help. Could you provide a small example that demonstrates the problem? Have you tried the example programs as well? If it freezes with those as well it'll be easier to help.

jonblack, 2017-08-10:
No response so closing. If you still need help with this, please re-open and comment.

closed 2017-08-10 by jonblack