jonblack/arduino-menusystem #55: pcd8544_nav example does not get compiled in atom

Npizza, 2017-07-19:
Hello i am trying to run  pcd8544_nav example in atom and it can't get build.I get this message
```
src/main.cpp:47:1: error: expected class-name before '{' token
{
^
src/main.cpp:63:40: error: 'BackMenuItem' has not been declared
virtual void render_back_menu_item(BackMenuItem const& menu_item) const
^ 
```

I think the problem is in the class MyRenderer but i dont know what to do

vogt31337, 2017-08-03:
I've stumbled across the same issue. Solution to the problem:
Don't use the add library function of the Arduino IDE it will install tag version 2.1.1, in which MenuSystem.h doesn't export MenuComponentRenderer class.

Instead download the master branch and install it in your library folder (like download zip and unzip it), rename from "arduino-menusystem-master" to "arduino-menusystem" and it should work.

@jonblack could you make a new tag version?

jonblack, 2017-08-03:
@vogt31337 thanks for finding this out. I'll make a new release soon.

jonblack, 2017-08-05:
Closing as this is fixed in the master and will be resolved when a release is made.

closed 2017-08-05 by jonblack