jonblack/arduino-menusystem #65: lcd_nav.ino doesn't compile

scls19fr, 2017-08-24:
Hello,

I try to compile this example with Arduino 1.8.2 and arduino-menusystem 2.1.1
https://raw.githubusercontent.com/jonblack/arduino-menusystem/master/examples/lcd_nav/lcd_nav.ino

but I get this error message:

```
Arduino : 1.8.2 (Mac OS X), Carte : "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

menu:26: error: expected class-name before '{' token
 class MyRenderer : public MenuComponentRenderer {
                                                 ^
menu:40: error: 'BackMenuItem' has not been declared
     void render_back_menu_item(BackMenuItem const& menu_item) const {
                                ^
menu:44: error: 'NumericMenuItem' has not been declared
     void render_numeric_menu_item(NumericMenuItem const& menu_item) const {
                                   ^
/Users/scls/Documents/Arduino/menu/menu.ino: In member function 'void MyRenderer::render(const Menu&) const':
menu:33: error: 'const class Menu' has no member named 'get_current_component'
         menu.get_current_component()->render(*this);
              ^
/Users/scls/Documents/Arduino/menu/menu.ino: In member function 'void MyRenderer::render_back_menu_item(const int&) const':
menu:41: error: request for member 'get_name' in 'menu_item', which is of non-class type 'const int'
         lcd.print(menu_item.get_name());
                             ^
/Users/scls/Documents/Arduino/menu/menu.ino: In member function 'void MyRenderer::render_numeric_menu_item(const int&) const':
menu:45: error: request for member 'get_name' in 'menu_item', which is of non-class type 'const int'
         lcd.print(menu_item.get_name());
                             ^
/Users/scls/Documents/Arduino/menu/menu.ino: At global scope:
menu:62: error: no matching function for call to 'MenuSystem::MenuSystem(MyRenderer&)'
 MenuSystem ms(my_renderer);
                          ^
/Users/scls/Documents/Arduino/menu/menu.ino:62:26: note: candidates are:
In file included from /Users/scls/Documents/Arduino/menu/menu.ino:11:0:
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:81:5: note: MenuSystem::MenuSystem()
     MenuSystem();
     ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:81:5: note:   candidate expects 0 arguments, 1 provided
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:78:7: note: constexpr MenuSystem::MenuSystem(const MenuSystem&)
 class MenuSystem
       ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:78:7: note:   no known conversion for argument 1 from 'MyRenderer' to 'const MenuSystem&'
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:78:7: note: constexpr MenuSystem::MenuSystem(MenuSystem&&)
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:78:7: note:   no known conversion for argument 1 from 'MyRenderer' to 'MenuSystem&&'
menu:63: error: no matching function for call to 'MenuItem::MenuItem(const char [24], void (*)(MenuComponent*))'
 MenuItem mm_mi1("Level 1 - Item 1 (Item)", &on_item1_selected);
                                                              ^
/Users/scls/Documents/Arduino/menu/menu.ino:63:62: note: candidates are:
In file included from /Users/scls/Documents/Arduino/menu/menu.ino:11:0:
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:34:5: note: MenuItem::MenuItem(const char*)
     MenuItem(const char* name);
     ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:34:5: note:   candidate expects 1 argument, 2 provided
Plusieurs bibliothèque trouvées pour "LiquidCrystal.h"
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note: constexpr MenuItem::MenuItem(const MenuItem&)
 class MenuItem : public MenuComponent
       ^
Utilisé : /Users/scls/Documents/Arduino/libraries/LiquidCrystal
Non utilisé : /Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note:   candidate expects 1 argument, 2 provided
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note: constexpr MenuItem::MenuItem(MenuItem&&)
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note:   candidate expects 1 argument, 2 provided
menu:64: error: no matching function for call to 'MenuItem::MenuItem(const char [24], void (*)(MenuComponent*))'
 MenuItem mm_mi2("Level 1 - Item 2 (Item)", &on_item2_selected);
                                                              ^
/Users/scls/Documents/Arduino/menu/menu.ino:64:62: note: candidates are:
In file included from /Users/scls/Documents/Arduino/menu/menu.ino:11:0:
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:34:5: note: MenuItem::MenuItem(const char*)
     MenuItem(const char* name);
     ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:34:5: note:   candidate expects 1 argument, 2 provided
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note: constexpr MenuItem::MenuItem(const MenuItem&)
 class MenuItem : public MenuComponent
       ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note:   candidate expects 1 argument, 2 provided
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note: constexpr MenuItem::MenuItem(MenuItem&&)
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note:   candidate expects 1 argument, 2 provided
menu:66: error: no matching function for call to 'MenuItem::MenuItem(const char [24], void (&)(MenuComponent*))'
 MenuItem mu1_mi1("Level 2 - Item 1 (Item)", on_item3_selected);
                                                              ^
/Users/scls/Documents/Arduino/menu/menu.ino:66:62: note: candidates are:
In file included from /Users/scls/Documents/Arduino/menu/menu.ino:11:0:
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:34:5: note: MenuItem::MenuItem(const char*)
     MenuItem(const char* name);
     ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:34:5: note:   candidate expects 1 argument, 2 provided
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note: constexpr MenuItem::MenuItem(const MenuItem&)
 class MenuItem : public MenuComponent
       ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note:   candidate expects 1 argument, 2 provided
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note: constexpr MenuItem::MenuItem(MenuItem&&)
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:31:7: note:   candidate expects 1 argument, 2 provided
/Users/scls/Documents/Arduino/menu/menu.ino: In function 'void serial_handler()':
menu:105: error: 'class MenuSystem' has no member named 'display'
                 ms.display();
                    ^
menu:109: error: 'class MenuSystem' has no member named 'display'
                 ms.display();
                    ^
menu:113: error: 'class MenuSystem' has no member named 'display'
                 ms.display();
                    ^
menu:117: error: 'class MenuSystem' has no member named 'display'
                 ms.display();
                    ^
/Users/scls/Documents/Arduino/menu/menu.ino: In function 'void setup()':
menu:137: error: 'class MenuSystem' has no member named 'get_root_menu'
     ms.get_root_menu().add_item(&mm_mi1);
        ^
menu:138: error: 'class MenuSystem' has no member named 'get_root_menu'
     ms.get_root_menu().add_item(&mm_mi2);
        ^
menu:139: error: 'class MenuSystem' has no member named 'get_root_menu'
     ms.get_root_menu().add_menu(&mu1);
        ^
menu:140: error: no matching function for call to 'Menu::add_item(MenuItem*)'
     mu1.add_item(&mu1_mi1);
                          ^
/Users/scls/Documents/Arduino/menu/menu.ino:140:26: note: candidate is:
In file included from /Users/scls/Documents/Arduino/menu/menu.ino:11:0:
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:57:10: note: void Menu::add_item(MenuItem*, void (*)(MenuItem*))
     void add_item(MenuItem* pItem, void (*on_select)(MenuItem*));
          ^
/Users/scls/Documents/Arduino/libraries/arduino-menusystem/MenuSystem.h:57:10: note:   candidate expects 2 arguments, 1 provided
menu:142: error: 'class MenuSystem' has no member named 'display'
     ms.display();
        ^
exit status 1
expected class-name before '{' token

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.
```

Any idea?

Maybe you should consider Continuous Integration?
http://docs.platformio.org/en/latest/ci/travis.html
https://travis-ci.org/jonblack/arduino-menusystem
and display a badge
[image]
to show if everything compile fine (and even pass unit tests... but that's an other story)

Kind regards

scls19fr, 2017-08-24:
After restarting IDE it works fine.
Sorry.
But CI is still something you should consider
Thanks

scls19fr, 2017-08-24:
In fact this problem occurs when copy/pasting this example code. When opening example directly in `/Users/scls/Documents/Arduino/libraries/arduino-menusystem/examples` it works fine.

jonblack, 2017-08-24:
When I paste the same example into the Arduino IDE it compiles fine. If I remove the `#include <MenuSystem.h>` line, I get the same errors as you, so I'm going to assume it's a copy and paste error.

Continuous integration would be nice but for such a small project with so few contributers and so few releases, it's a little overkills. Perhaps when I get around to adding tests it'd be more useful.

closed 2017-08-24 by jonblack