jonblack/arduino-menusystem #66: Add unit tests

jonblack, 2017-08-24:
There are no tests at the moment which makes refactoring difficult. The tests would ideally run on the host rather than on an Arduino to make continuous integration easier.

scls19fr, 2017-08-24:
Ensure that code can be compiled successfully (which is currently the case) could be a first step.
So when a PR is send it will ensure it compiles but you are right unit tests are a good practice.

scls19fr, 2017-08-24:
Here is a first example of a `.travis.yml` file

```
language: python
python:
    - "2.7"

# Cache PlatformIO packages using Travis CI container-based infrastructure
sudo: false
cache:
    directories:
        - "~/.platformio"

env:
    - PLATFORMIO_CI_SRC=examples/lcd_nav
    - PLATFORMIO_CI_SRC=examples/current_item
    - PLATFORMIO_CI_SRC=examples/current_menu
    - PLATFORMIO_CI_SRC=examples/lcd_nav
    - PLATFORMIO_CI_SRC=examples/led_matrix
    - PLATFORMIO_CI_SRC=examples/led_matrix_animated
    - PLATFORMIO_CI_SRC=examples/pcd8544_nav
    - PLATFORMIO_CI_SRC=examples/serial_nav

install:
    - pip install -U platformio

    #
    # Libraries from PlatformIO Library Registry:
    # missing ht1632c for led_matrix
    # missing ht1632c for led_matrix_animated
    - platformio lib -g install Adafruit_GFX  # for pcd8544_nav

script:
    #- platformio ci --board=uno --board=teensy31 --board=due --lib="."
    - platformio ci --board=uno --lib="."
```

I'm still facing several issues.

Despite Adafruit_GFX is installed using

    $ platformio lib -g install Adafruit_GFX

(for pcd8544_nav example)

running locally this example using:

```
$ export PLATFORMIO_CI_SRC=examples/pcd8544_nav
$ platformio ci --board=uno --lib="."
*********************************************************************************************************************************************
Obsolete PIO Core v3.4.1 is used (previous was 3.5.0a4)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*********************************************************************************************************************************************
The next files/directories have been created in /private/var/folders/z2/1vydt2v5213482bn6yry5vn40000gn/T/tmpPprtUy
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
[Thu Aug 24 16:24:25 2017] Processing uno (platform: atmelavr; board: uno; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting pcd8544_nav.ino
Collected 27 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <arduino-menusystem> v3.0.0
|-- <SPI> v1.0
Compiling .pioenvs/uno/src/pcd8544_nav.ino.o
Archiving .pioenvs/uno/libFrameworkArduinoVariant.a
Compiling .pioenvs/uno/FrameworkArduino/CDC.o
Indexing .pioenvs/uno/libFrameworkArduinoVariant.a
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial.o
/private/var/folders/z2/1vydt2v5213482bn6yry5vn40000gn/T/tmpPprtUy/src/pcd8544_nav.ino:13:26: fatal error: Adafruit_GFX.h: No such file or directory
#include <Adafruit_GFX.h>
^
compilation terminated.
Compiling .pioenvs/uno/FrameworkArduino/HardwareSerial0.o
*** [.pioenvs/uno/src/pcd8544_nav.ino.o] Error 1
========================================================= [ERROR] Took 1.70 seconds =========================================================
```

raise error (`Adafruit_GFX.h` is not found)

Build also fails for `led_matrix` and `led_matrix_animated`.
But that's more understandable (because ht1632c is missing) and I didn't find an easily installable version https://github.com/PaulStoffregen/ht1632c/issues/1 (an Arduino or PlatformIO package).

Others examples can be compiled on Travis side without any problem.