jonblack/arduino-menusystem #82: Methods (function)

renejeanmercier, 2019-05-28:
Hi Mr. Black,

Firstly, thank you for that unique and well thought menu system. The fact that each menu item can call a function to adapt to the user needs and all sorts of displays makes your menu system universally usable.

Jon, in the documentation at https://jonblack.me/arduino-menu-system-library/, you refer to two (2) method : ms.get_num_menu_items(); // get the number of menu items in the current menu and ms.get_cur_menu_item(); // get the current menu item number in the current menu.

These methods are not present in the actual library. Is there a reason ?

Regards,

Rene-J. Mercier

jonblack, 2019-05-28:
The documentation at that link is out of date. The API has changed. Take a look at the examples for how to interact via the new API.

closed 2019-05-28 by jonblack

renejeanmercier, 2019-05-31:
Hi Jon,

Thank you for your quick reply to my previous email.

I have 6 main menu items in my application and each sub-menu has a
different number of items (see below), but I can only display 3 items at a
time on the TFT display due to other information displayed in an other area
of the TFT.

Therefore I need to scroll the menu items up and down in a kind of
windowing approach : say I have items 1, 2 and 3 displayed. If the focus is
already on item 3 and I turn the rotary encoder down one step, I need to
display items 2, 3, and 4, an so on and vis-versa.

The get_num_menu_items() and get_cur_menu_item() where perfect for such
functionality. Before I undertake to write such functions outside of the
"menu system" (in my application) I was wondering if you had replaced these
functions with equivalent functions in your actual version. Or you might
want to help me (your C++ caliber is in the top professional, mine at the
bottom of the barrel ...hold man in the 70 years of age trying to catch-up
:-)) in suggesting an approach to follow.

Thank you so much for your help and understanding.

Regards,

Rene-J Mercier

email : renejeanmercier@gmail.com

============================
void on_item10_selected(MenuComponent* p_menu_component); // Generator One
void on_item20_selected(MenuComponent* p_menu_component); // Generator Two
void on_item30_selected(MenuComponent* p_menu_component); // Generator Three
void on_item40_selected(MenuComponent* p_menu_component); // Language
void on_item50_selected(MenuComponent* p_menu_component); // Diagnostics
void on_item60_selected(MenuComponent* p_menu_component); // Communications

void on_item11_selected(MenuComponent* p_menu_component); // Freq. step.
value
void on_item12_selected(MenuComponent* p_menu_component); // Frequency
increment
void on_item13_selected(MenuComponent* p_menu_component); // Phase
adjustment
void on_item14_selected(MenuComponent* p_menu_component); // Return to main
menu

void on_item21_selected(MenuComponent* p_menu_component); // Freq. step
value
void on_item22_selected(MenuComponent* p_menu_component); // Frequency
increment
void on_item23_selected(MenuComponent* p_menu_component); // Phase
adjustment
void on_item24_selected(MenuComponent* p_menu_component); // Return to main
menu

void on_item31_selected(MenuComponent* p_menu_component); // Freq. step
value
void on_item32_selected(MenuComponent* p_menu_component); // Frequency
increment
void on_item33_selected(MenuComponent* p_menu_component); // Phase
adjustment
void on_item34_selected(MenuComponent* p_menu_component); // Return to main
menu

void on_item41_selected(MenuComponent* p_menu_component); // Language 1 --
French
void on_item42_selected(MenuComponent* p_menu_component); // Language 2 --
English
void on_item43_selected(MenuComponent* p_menu_component); // Return to main
menu

void on_item51_selected(MenuComponent* p_menu_component); // Diagnostics 1
-- SI5351A
void on_item52_selected(MenuComponent* p_menu_component); // Diagnostics 2
-- I2C
void on_item53_selected(MenuComponent* p_menu_component); // Diagnostics 2
-- Rotary Encoder
void on_item54_selected(MenuComponent* p_menu_component); // Return to main
menu

void on_item61_selected(MenuComponent* p_menu_component); // Serial
Communications -- Control
void on_item62_selected(MenuComponent* p_menu_component); // Return to main
menu

willie68, 2019-05-31:
I have a similar problem. I have an 16x2 LCD so i can only display 2 lines of the menu. So i display in the first row the parent menu item and in the second row the active menu itemn or value. With the encoder i can rotate thru all menu items, a click just activates the selected item. If it's a menu, it opens it, if it's a value, the value editor starts. In every submenu i have a back menu item,  just to go one level higher.

[settings.zip](https://github.com/jonblack/arduino-menusystem/files/3240799/settings.zip)